OLIMEX/ESP32-EVB #1: Здравейте!

me-no-dev, 2017-02-14:
Здравейте!
Казвам се Христо Гочков, работя за Еспресиф и в частност съм отговорен за Ардуино средата за ЕСП32. Понеже сме си сънародници :) реших да ви пиша и да ви кажа че ща се радвам ако можем да сме си полезни, пък и да ви оставя контакт :) Ако имате въпроси относно контролерите, не се колебайте да ги зададете.

ESP32DE, 2017-02-15:
Fine! You are welcome Christo ;-)

best wishes
rudi ;-)

TsvetanUsunov, 2017-02-15:
Здрасти, прати един имейл на info @ olimex com

closed 2017-07-19 by DanKoloff

stankow, 2018-09-04:
Малко късно попадам на тази поща, чак след година. Работим с ESP32, много сме доволни ! Ценим много вашият принос ! Прехвърляме наш код от предишни Ардуино проекти и вашите библиотеки за ESP32 добре ни идват !  Поздрави, успех !

me-no-dev, 2018-09-04:
Нищо :) тя офертата си е валидна и сега