OLIMEX/ESP32-EVB #5: Error in naming on schematics - IR block GPIO39 misnamed (lack of letter) :)

Netoperz, 2017-07-04:
as in title.

:)

DanKoloff, 2017-07-05:
I don't understand what do you mean. What is wrong and how should it be corrected?

DanKoloff, 2017-08-17:
I would close this since I believe what @Netoperz meant is that GPI36 should be named GPIO36; which is not very correct since it can be only input (it can't be output); so GPI is a more fitting abbreviation.

closed 2017-08-17 by DanKoloff

Netoperz, 2017-08-24:
Yes, the pin is input only in arduino ide. but can be innput/outp[ut if using driver on espressif ide. so naming it GPI is fine because normally without any magic it is input only.

Issu should be closed.

DanKoloff, 2017-08-25:
I am not happy with your explanation which might perplex someone reading this. The input/output availability has nothing to do with Arduino IDE, or ESP-IDF. In order to clarify I would cite version 1.7 of ESP32's datasheet, page 36: 

"GPIO pins 34-39 are input-only. These pins do not feature an output driver or internal pullup/pull-down
circuitry. The pin names are: SENSOR_VP (GPIO36), SENSOR_CAPP (GPIO37), SENSOR_CAPN (GPIO38), SENSOR_VN (GPIO39), VDET_1 (GPIO34), VDET_2 (GPIO35)."