OLIMEX/ESP32-EVB #6: USB-UART Circuit

Ayush123Singh, 2017-07-11:
Hi,
I am new to ESP32 and i am having some doubt in USB-UART circuit, what are the uses of DTR, DSR, RTS, CTS pin in CP2102.
If i have asked some obviously simple question then sorry for inconvenience but little help will be very much appreciated. Thanks.

StefanMavrodiev, 2017-07-12:
We are using CH340T, not CP2102.
CTS and DSR pins are not used.

On the schematic there is table about the work of DTR and RTS pins.
In short:
DTR = 1, RTS = 0 - the board is in reset
DTR = 0, RTS = 1 - GPIO0 is pulled low

carhero, 2017-07-12:
Are you having trouble to connect ESP32-EVB with windows?
If so please refer below.
https://github.com/OLIMEX/ESP32-EVB/issues/8

here is direct link for CH340T
http://www.winchiphead.com/download/CH341/CH341SER.ZIP

closed 2017-08-17 by DanKoloff