OLIMEX/ESP32-EVB #7: SD card is not working!

pbecchi, 2017-07-11:
There is no documentation or example, but in EVB board rev B , the SD adapter is showing a SPI connection but SS pin is not connected to any GPio and is pulled up!
To allow communication from the SD, SS should be connected and/or pulled down.

Any other way to make it work?

StefanMavrodiev, 2017-07-12:
SS should be connected if you're using SD card in SPI mode, which is unnecessary. SD protocol needs only 3 pins: CMD, DATA0 and CLK.

You can check Physical Layer Simplified Specification here:
https://www.sdcard.org/downloads/pls/index.html

We will upload SD-card example later today.

StefanMavrodiev, 2017-07-12:
You can see working example here:
https://github.com/OLIMEX/ESP32-EVB/tree/master/SOFTWARE/sd_card

Here is the example output:

```
I (967) example: Initializing SD card
Name: SL08G
Type: SDHC/SDXC
Speed: default speed
Size: 7580MB
CSD: ver=1, sector_size=512, capacity=15523840 read_bl_len=9
SCR: sd_spec=2, bus_width=5
I (1137) example: Opening file
I (1247) example: File written
I (1317) example: Renaming file
I (1347) example: Reading file
I (1357) example: Read from file: 'Hello SL08G!'
I (1357) example: Card unmounted
```

pbecchi, 2017-07-12:
I have tested your example : it work only one time over ten trial......don't ask me why.
Arduino-core SdMMc example never work and it should!
May be you closed the issue too soon!

StefanMavrodiev, 2017-07-12:
> it work only one time over ten trial......don't ask me why.
You don't know or you won't tell us?

If you do the same change in the arduino library, example should work:

https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/src/SD_MMC.cpp
```
host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;
//host.flags = SDMMC_HOST_FLAG_1BIT; //use 1-line SD mode
```
to

```
host.max_freq_khz = SDMMC_FREQ_PROBING;
host.flags = SDMMC_HOST_FLAG_1BIT; //use 1-line SD mode
```

pbecchi, 2017-07-12:
Yes thanks
It is what I did , in my case it didn't work! May be you have to try this yourself.
Major problem anyhow is reliability : your code it doesn't work all the time.
I doubt may be due to the default pin left unconnected and may be used by other applications.

StefanMavrodiev, 2017-07-12:
Here is my output.

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
SD_MMC Card Type: SDHC
SD_MMC Card Size: 7580MB
                        Listing directory: /
                                              FILE: /test.txt  SIZE: 0
  FILE: /foo.txt  SIZE: 13
Creating Dir: /mydir
                    Dir created
Listing directory: /
                      FILE: /test.txt  SIZE: 0
  FILE: /foo.txt  SIZE: 13
  DIR : /mydir
Removing Dir: /mydir
                    Dir removed
Listing directory: /
                      FILE: /test.txt  SIZE: 0
  FILE: /foo.txt  SIZE: 13
Writing file: /hello.txt
                        File written
Appending to file: /hello.txt
                             Message appended
Reading file: /hello.txt
                        Read from file: Hello World!
                                                    Deleting file: /foo.txt
                                                                           File deleted
Renaming file /hello.txt to /foo.txt
                                    File renamed
Reading file: /foo.txt
                      Read from file: Hello World!
                                                  0 bytes read for 0 ms
                                                                       1048576 bytes written for 25351 ms

```

Maybe there is something wrong with the SD-card?

pbecchi, 2017-07-12:
The main problem is in HW.
I guess is probably a contact from card holder and SD. Moving the SD card goes and fail.
I tried also with core SdMMc library with similar results : could be the holder of the SD but I tried with several SD.
I had also a problem of SD non written by the SW (with no error) probably because it was a Fat32 SD.

StefanMavrodiev, 2017-07-12:
Could you try several tests, while you press the SD card towards the board?

pbecchi, 2017-07-12:
Done several test, unfortunately I have only 2 SD formatted FAT16 : 
for booth moving SD sideways (perpendicular to insertion movement) create bad or good contact.
This may be due to an holder too large or due to two narrow SD.

closed 2017-08-25 by TsvetanUsunov

quikue, 2019-02-11:
Sorry,
:)
I have the problem of I can use SPI with SD_MMC,
I understand that the pinout is:
MISO 02
MOSI 15
CLK    14
CHIP SELECT CS0 14
But that also need a SPI with four wires.

Can you help me?

Thanks!

quikue, 2019-02-11:
Sorry,
:)
I have the problem of I can't use SPI with SD_MMC,
I understand that the pinout is:
MISO 02
MOSI 15
CLK 14
CHIP SELECT CS0 14
But that also need a SPI with four wires.

Can you help me?

Thanks!

bioshock2k, 2020-10-24:
Hi, did anyone had it working with the Arduino IDE?

teson2000, 2023-08-24:
SD_CARD working in Arduino IDE when changing library from SD to MMC.
(This fixes the 4-bit vs 1-bit problem)
Likewise, any references to SD needs to be changed to MMC. Working example:

```
#include "SD_MMC.h"
#include "SPI.h"

void initSDCard(){
  if(!SD_MMC.begin("/sdcard",true)) {
    Serial.println("Card Mount Failed");
    return;
  }
  uint8_t cardType = SD_MMC.cardType();

  if(cardType == CARD_NONE){
    Serial.println("No SD card attached");
    return;
  }

  Serial.print("SD Card Type: ");
  if(cardType == CARD_MMC){
    Serial.println("MMC");
  } else if(cardType == CARD_SD){
    Serial.println("SDSC");
  } else if(cardType == CARD_SDHC){
    Serial.println("SDHC");
  } else {
    Serial.println("UNKNOWN");
  }
  uint64_t cardSize = SD_MMC.cardSize() / (1024 * 1024);
  Serial.printf("SD Card Size: %lluMB\n", cardSize);
}

void setup() {
  delay(1000);
  Serial.begin(115200);
  initSDCard();
}  

void loop() {
  
}
```

gjt211, 2023-09-15:
@teson2000, thanks kindly - I have been trying to get this to work for hours. The SD_MMC example works for me.