OLIMEX/ESP32-EVB #10: Flash / monitor failure

jonface, 2017-08-09:
I am getting the following errors when trying to flash and use monitor, unit is brand new.

The default linux ch341 driver gave me input/output error when trying to open the serial port. I've compiled and inserted the ch34x driver from your website which at least allows me to open the serial port.

Any ideas? Broken unit?

Thanks

```
/hello_world$ make flash
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.1-beta1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
/home/jmasters/Downloads/xtensa-esp32-elf/esp-idf/components/esptool_py/Makefile.projbuild:52: recipe for target 'flash' failed
make: *** [flash] Error 2
```

```
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

... This goes on forever
```

```
Linux vmxubuntu16 4.4.0-45-generic #66-Ubuntu SMP Wed Oct 19 14:12:37 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

`chmod 777 on /dev/ttyUSB0`

jonface, 2017-08-09:
I have also tried from an un powered USB hub and powered. Also external 5v.

DanKoloff, 2017-08-14:
The re-occurring message is not surprising ("flash read err, 1000"). The board comes with empty flash. 

It is more interesting why the programming fails. Probably very high baud rate (try with lower) or bad USB port (try with another USB port).

jonface, 2017-08-14:
OK, I setup the environment using the windows UART drivers, Arduino IDE and it is flashing fine @ 115200

I think, at a guess, it is the linux UART driver, even at 9600 I get nothing. I might try Wireshark at some point and check, if anything, what the USB port is doing.

The main thing is the hardware is working, feel free to close this.

Thanks

DanKoloff, 2017-08-14:
By default Linux should come with properly working ch34x driver. The driver that we published is only for old kernels older than version 3.14. Your ubuntu has kernel 4.4.0+ and doesn't need this driver.

jonface, 2017-08-16:
I've tried again, with the built in HL-340 driver,

`1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter`

monitor is working, I understand what you meant previously now about the error, as I can see the serial output fine from the application, but flashing still doesn't work (tried 9600/115200)

```
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.1-beta1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
/home/jmasters/Downloads/xtensa-esp32-elf/esp-idf/components/esptool_py/Makefile.projbuild:52: recipe for target 'flash' failed
make: *** [flash] Error 2
```

I'm assuming this isn't technically a problem with the ESP32-EVB so I should really head over to the ESP github?

Thanks

DanKoloff, 2017-08-16:
Since you managed to program it via the Arduino, I'd guess the problem is not in the hardware.

jonface, 2017-08-16:
Closing as problem not specific to Olimex ESP32-EVB.

Thanks.

closed 2017-08-16 by jonface