OLIMEX/ESP32-EVB #11: Emac rx buff err in ETH example

gjongenelen, 2017-09-28:
Hello,

I bought a ESP21-EVB Rev.C  board and am trying to run the ETH_PHY_demo program. Connection to the PHY is established, but I'm getting the following error:
`E (137287) emac: emac rx buf err!!`

Is this a result of using Rev.B examples on a Rev.C board?

DanKoloff, 2017-09-29:
The software example was written for older version of ESP-IDF, it needs to be updated, for now follow the advice I posted here until we fix it: https://github.com/OLIMEX/ESP32-GATEWAY/issues/3

gjongenelen, 2017-10-09:
Thanks, it works

jovermann, 2017-12-13:
I see the same issue with the Olimex ETH_PHY_demo (also using a Rev C board). I can also confirm that the workaround pointed to by the link above works.

Concretely, the workaround is:
1. Use the ethernet example provided with the esp-idf. Do not use the ETH_PHY_demo at all.
2. Use these settings in make menuconfig:
CONFIG_PHY_TLK110=
CONFIG_PHY_LAN8720=y
CONFIG_PHY_ADDRESS=0
CONFIG_PHY_CLOCK_GPIO0_IN=y
CONFIG_PHY_CLOCK_GPIO0_OUT=
CONFIG_PHY_CLOCK_GPIO16_OUT=
CONFIG_PHY_CLOCK_GPIO17_OUT=
CONFIG_PHY_CLOCK_MODE=0
CONFIG_PHY_USE_POWER_PIN=
CONFIG_PHY_SMI_MDC_PIN=23
CONFIG_PHY_SMI_MDIO_PIN=18

DanKoloff, 2018-03-27:
The examples had been updated and should work properly with the latest ESP-IDF.

closed 2018-06-07 by gjongenelen