OLIMEX/ESP32-EVB #13: ESP32-EVB Non-Evaluation board

Elkrival, 2017-12-18:
Hello Olimex,

Is there a production ready version of this board?

Best,

Ivan

DanKoloff, 2017-12-18:
What does this mean? Like stripped down version of ESP32-EVB with less features?

Elkrival, 2017-12-18:
On the Olimex website there's this notice.

 Evaluation Board/Kit Important Notice

OLIMEX Ltd provides the enclosed product(s) under the following conditions: 

This evaluation board/kit/module is intended for use for ENGINEERING DEVELOPMENT, DEMONSTRATION OR EVALUATION PURPOSES ONLY and is not considered by OLIMEX Ltd to be finished end-product fit for general consumer use. Persons handling the product must have electronics training and observe good engineering practice standards. As such the goods being provided are not intended to be complete in terms of required design-, marketing-, and/or manufacturing related protective considerations, including product safety and environmental measures typically found in the products that incorporate such semiconductor components or circuit boards. This evaluation board/kit/module does not fall within the scope of the European Union directives regarding electromagnetic compatibility, FCC, CE or UL and therefore may not meet the technical requirements of these directives or other related documents. 

The user assumes all responsibility and liability for proper and safe handling of the goods. Further the user indemnifies OLIMEX from all claims arising from the handling or use of the goods. Due to the open construction of the product, it's the user responsibility to take any and all appropriate precautions with regard to electrostatic discharge, conducted and radiated emissions. 

EXCEPT TO THE EXTENT OF THE INDEMNITY SET FORTH ABOVE NEITHER PARTY SHALL BE LIABLE TO THE OTHER FOR ANY INDIRECT SPECIAL INCIDENTAL OR CONSEQUENTIAL DAMAGES.

I wanted to know if there's a model of this board that is not only for development.

DanKoloff, 2017-12-18:
This is the only version of the ESP32-EVB board (we have a variant called ESP32-GATEWAY). We have the exact same notice under ALL of our products. Feel free to use the product for professional purposes, a lot of people embed our boards in their designs. There are reasons for the notice, but I bet you can find why it is there yourself.

Elkrival, 2017-12-18:
Appreciate the help and Thank you for the quick response!

closed 2017-12-18 by DanKoloff