OLIMEX/ESP32-EVB #14: ETH-PHY  demo doesn't work

hexanet67, 2018-01-25:
Hi,

I'm trying ETH_PHY_demo on Olimex ESP32-EVB. No problem with compilation, but when flash compiled firmware, yellow led on ETH connector start flashing and I can see thi in terminal log window: 

`I (257) cpu_start: Pro cpu start user code
I (275) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (277) emac: mac version 1137a
I (277) emac: emac start !!!

I (277) emac: emac reset done
I (287) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
E (1297) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
E (2297) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff
E (3297) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
E (4297) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff
E (5297) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
E (6297) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff
E (7297) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
`
I have also tried with external power supply 5V/2A but results are the same.
Any idea how to get ethernet working ?

Regards,

    Jan

DanKoloff, 2018-01-26:
Take a look here: https://github.com/OLIMEX/ESP32-EVB/issues/11

hexanet67, 2018-01-26:
Hi,
thank you for a quick reply.
I did what is described in 'Emac rx buff err in ETH example #11' and I can confirm that it worked for me.
I was trying original ethernet example in esp-idf before, but probably didn't set all configuration options correctly.
Regards,
    Jan

closed 2018-01-26 by DanKoloff