OLIMEX/ESP32-EVB #18: ethernet can't be connected with ethernet

shubham7946, 2018-07-26:
This type of error is coming while connecting with LAN to olimex.My connected network is of DHCP type.

E (1323) emac: Timed out waiting for PHY register 0x2 to have value 0x2000 (mask 0xffff). Current value 0xffff
E (2323) emac: Timed out waiting for PHY register 0x3 to have value 0xa210 (mask 0xfff0). Current value 0xffff
E (3323) emac: Timed out waiting for PHY register 0x2 to have value 0x2000 (mask 0xffff). Current value 0xffff
E (4323) emac: Timed out waiting for PHY register 0x3 to have value 0xa210 (mask 0xfff0). Current value 0xffff

Kindly suggest me solution.

DanKoloff, 2018-07-26:
Did you try using exactly this demo: https://github.com/OLIMEX/ESP32-EVB/tree/master/SOFTWARE/ETH_PHY_demo ?

maqtech, 2018-10-03:
Hi I tried to use the demo with a REV C, but the board does not connect to ethernet, could you please support me!

DanKoloff, 2018-10-04:
What exactly happens? Is there some error message or where does it stop?

DanKoloff, 2020-07-17:
Closing due to lack of reply and details. If the problem remains and new details emerge I'd reopen it.

closed 2020-07-17 by DanKoloff