OLIMEX/ESP32-EVB #21: Working on examples for Arduino

RevoluPowered, 2019-06-03:
I'm working on Arduino code examples for this board, do you mind if I submit a PR with them?

It will include the button usage, relay control, can interface, 1-wire demo for (DS18B20 sensors).

TsvetanUsunov, 2019-06-04:
no problem, let us review

RevoluPowered, 2019-06-04:
Cool will do, I will fork and merge later on when I'm back from work.

P.S. Really liking the EVB

RevoluPowered, 2019-06-07:
@TsvetanUsunov Still working on this but having some issues with the SD library from arduino-esp32. 

Reading from SD MMC card slot with 1 wire bus interface takes 7 seconds to read 1 MB, even when using optimal settings but it does function.

Anything you know which can possibly help improve this?

I will probably do performance read/write tests with the ESP32 SDK

RevoluPowered, 2019-06-08:
https://github.com/espressif/arduino-esp32/issues/1611

Digging is available, appears to not be an issue with the evb, seems to be related to arduino-esp32 library as confirmed by my tests.

wimmmb, 2019-08-02:
Hi, i get the watering example not compiling, i placed the WebPage.c in data, uploaded it with the ESP32 sketch data upload tool and try to compile, get the error:
sketch\ESP32_Watering.ino.cpp.o:(.literal._Z13Print_WebpageR10WiFiClient+0x4): undefined reference to `WebPage'

Seems the external doesn work somehow:
extern const char WebPage [168][80];

Thanks, Wim

DanKoloff, 2020-07-17:
Closing due to inactivity. If you are still interested please comment I would re-open it.

closed 2020-07-17 by DanKoloff