OLIMEX/ESP32-EVB #22: Program does not start correctly with GPIO4 on high.

Nestornp, 2019-08-22:
I need the UART1 for my project, with GPIO4 for U1TX and GPIO36 for U1RX. Uart on ESP32 works at 3.3v but the destiny UART target works at 2.7v, so i'm using a Logic Level Converter (the standar TXS0108E). The problem is that with this converter, the gpio4 is high when you connect the power and the olimex esp32 does not start the program. I do not understand why. It is necessary to press the reset button on the olimex 32 to start the program correctly (When you press the reset, the gpio4 is set to 0 V and the program starts correctly). This is a serious problem for me, on my application I need to turn the olimex on and off without having to press the reset button. 
Any solution?

DanKoloff, 2019-08-22:
Look at the "Internal Bootstrapping Resistors" table GPIO4 is present and has a Pull-Down. Maybe this bugs the converter and the logic.

You can use one of the other pins at the UEXT connector instead of GPIO4 - Like all peripherals, the pins for the UARTs can be logically mapped to any of the available pins on the ESP32.

For example, try using GPIO16 (pin #6 from the UEXT); of course remember to change it in the code.

closed 2019-08-23 by Nestornp

nagyrobi, 2021-12-28:
For reference, from: https://www.olimex.com/forum/index.php?topic=8093.0
_It seems like adding 1K pull down resistor between GPIO4 and GND fixes the issue._