OLIMEX/ESP32-EVB #24: Use master branch of ESP-IDF (V4.x)

GISED-Link, 2019-12-02:
Hi,

We are currently running your demo code with an opcua server (see https://github.com/cmbahadir/opcua-esp32). To run your demo, we need ESP-IDF V3.3. That was the case 6 month ago. Is the project compatible with the new release of the ESP-IDF now?

At the moment, to update to the version 4.x of ESP-IDF, we need to rewrite the ETH part. If it is not done yet, we can do it. But it is possible to contribute on this git?

Thank you for the greate work until now 👍

DanKoloff, 2019-12-03:
We haven't updated our examples for ESP-IDF version 4.x since the official tags are still marked as either "dev" or "beta". There is still no stable release of ESP-IDF version 4.x. As soon as there is a stable tag we would start working on updating our examples. This was also announced here: https://github.com/OLIMEX/ESP32-POE/issues/7#issuecomment-523761856

The edit itself shouldn't be very difficult, a couple of paths are different. The last two posts of the issue thread that I linked above contain some instructions on getting the Ethernet working. These can be a bit different for ESP32-EVB (since it uses external clock generator, while all other Olimex ESP32 designs use the internal clock).

GISED-Link, 2019-12-04:
Thank you for the reply. I have some trouble between lwip and opcua layers (lwip and open62541) and I am not able to see what is happening. The guy who has ported open62541 to the esp32 use the latest V4.x of the ESP-IDF and has something that work quite properly. I really want to see if an update can fix the issue I have.

DanKoloff, 2020-07-17:
We published an Ethernet demo for version 4.0 here now: https://github.com/OLIMEX/ESP32-EVB/tree/master/SOFTWARE/ESP32-EVB_Ethernet_v4.0

closed 2020-07-17 by DanKoloff