OLIMEX/ESP32-EVB #25: Kicad issues

johanliebs, 2019-12-13:
Downloaded the Kicad files to have a look and I am getting errors opening the project, PCB layout and schematic files for both revisions D & F.

Project error: file '/home/johan/...Rev-F/NB-IoT-DevKit_RevB.pro',line 7:'=' expected

Ubuntu 18.04
Kicad 5.1.15 (also 5.02 before upgrade)

Thanks!

DanKoloff, 2019-12-16:
Mmmm why does it say "Project error: file '...NB-IoT-DevKit_RevB.pro',line 7:'=' expected" - did you try to open NB-IoT-DevKit_RevB.pro project?

I assume you checked out something different there is no such file here: https://github.com/OLIMEX/ESP32-EVB/tree/master/HARDWARE/REV-F

johanliebs, 2019-12-16:
At first I downloaded the ZIP file and extracted it to : /home/johan/kicad/olimex iot dev kit/Rev-F/. When I open the *.pro file:

[image]

Second attempt is manually downloading each file from Rev-F (right click - save link as) to : /home/johan/kicad/olimex iot dev kit/Rev-F_manual_download/.  Same errors as the zip file.

[image]

Also when I try to open any other file like the schematic, I get more errors. The 1st error:
[image]

But if I go to https://github.com/RevoluPowered/ESP32-EVB and do a ZIP download of Rev-D, Kicad opens the project just file without any errors.

[image]

DanKoloff, 2019-12-17:
Can the issue be related to the path that you used to store the design files and which contains 3 x "space" character, namely "olimex iot dev kit". One thing I've learned is that Linux doesn't like space character, let alone three of them.

If you look at the path of the successfully opening revision D files - there is no space.

Re-downloaded and tested the design files on several machines here. No such issues here. And if I can't replicate it I can't fix it. 

It doesn't seem like a problem with our design sources, not only because we have no such issue but also no one else reported something similar and these files had been out for some time. Aside from placing the project in path without "space" character I'd also suggest, that it might be some specific issue with the KiCAD release for your operating system. I'd recommend re-installing KiCAD and maybe trying latest nightly build for your system. If the problem persists report it to the KiCAD team.

johanliebs, 2019-12-18:
Removed all the spaces, made it simple, still the same error : 
[image]

johanliebs, 2019-12-18:
I did another test with Kicad  4.0.6 on Mageia 6, which works fine without errors. Will try to re-install Kicad on Ubuntu 18.04

johanliebs, 2019-12-18:
Same as before after un-installing and re-installing the latest version of Kicad on Ubuntu 18.04.... 

But if I go to https://github.com/RevoluPowered/ESP32-EVB and do a ZIP download of Rev-D, Kicad opens the project just file without any errors. Why would Rev-D work fine?

TsvetanUsunov, 2019-12-18:
Hi
if you open the PCB file with text editor you will read this:
(kicad_pcb (version 20171130) (host pcbnew 6.0.0-rc1-unknown-0cca1c6~65~ubuntu16.04.1)
which means this file is made with KiCad PCB 6.0.0-rc1-unknown-0cca1c6~65~ubuntu16.04.1
you have to install same version of KiCad (hint read how to install the nightly builds) to may open and work with the file.
You try to open files made with 6.0 with KiCad 4 or 5 which is not possible.

closed 2019-12-18 by TsvetanUsunov