OLIMEX/ESP32-EVB #29: Visual datasheet / hookup guides?

wolfspyre, 2020-07-27:
Hihi!

Thanks for updating the docs to incl rev. I of the board. 
I reviewed the site/ this project, and any links I could find, but have yet to locate a diagram similar to the way sparkfun describes their boards: 
take their esp32wroom for example: 
https://www.sparkfun.com/products/15663
if you look at the documents tab on  that page:
they make a graphical datasheet for the board:
https://cdn.sparkfun.com/assets/learn_tutorials/8/5/2/ESP32ThingPlusV20.pdf

and then do a 'hookup guide' 
https://learn.sparkfun.com/tutorials/esp32-thing-plus-hookup-guide?_ga=2.171677728.1493458959.1595877428-808389143.1595877428

that covers the nuance of that board.... it sorta serves as a collation point for information relating to that board. 

is there such a place for y'alls products and I'm just.... ignorant?

DanKoloff, 2020-07-28:
Thanks for the feedback. This is definitely something we are looking into creating. We will see about ESP32-EVB, we have similar pinouts for the rest of the boards, for example:

https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/resources/ESP32-GATEWAY-GPIOs-Rev.F-up.pdf
https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/resources/ESP32-POE-GPIO.png
https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/resources/ESP32-POE-ISO-GPIO.png

wolfspyre, 2020-07-31:
it appears riot-os has some of this info laid out for ya:

https://riot-os.org/api/group__boards__esp32__olimex-esp32-evb.html

closed 2020-10-13 by DanKoloff