OLIMEX/ESP32-EVB #30: Ethernet restart

iBobik, 2021-03-15:
Hello,
I sometimes have issues with Ethernet stability (probably due power stagility or interferences). This could be easilly fixed by a restart of whole board (SW restart on ESP is not enough). Powering off or short circuit 3.3V with GND restarted it well enough. But I need to do it automatically.

So it seems like I need to restart Ethernet controller. Could I somehow restart it from ESP by it's GPIOs?

DanKoloff, 2021-03-16:
"short circuit 3.3V with GND" is not good idea... better remove the 3.3V instead of short-circuiting the board...

The reset signal that is connected to the reset button is ESP_EN, it can be found on pin #33 of the EXT1 header. To reset the board you need to keep ESP_EN in low position and then release it back up. It is risky to connect it to ESP32's GPIO tho, you might shoot yourself in the foot.

iBobik, 2021-03-16:
Thank you for reply.

Will be safe to ground ESP_EN by a transistor controlled by some GPIO?
Which transistor shoudl work?

Honza Pobořil

DanKoloff, 2021-03-17:
First test directly with a free GPIO, place 1uF capacitor on the wire. It is worth testing the simple setup first.

iBobik, 2021-04-14:
Hello,
I've tried to ground ESP_EN by 1-4 0,95uF capacitors, but restart not happened.

iBobik, 2021-04-14:
Here someone recommended to just wire GPIO and RST:
https://www.researchgate.net/post/How-to-reboot-reset-ESP32-without-pressing-the-reboot-button-on-the-board

What could happen?

DanKoloff, 2021-04-30:
Well I wrote the same: "First test directly with a free GPIO, place 1uF capacitor on the wire. It is worth testing the simple setup first."

iBobik, 2021-04-30:
Sorry, I am not sure I understand you, so I draw it :-)

Here is what I tried:
[image]

iBobik, 2021-04-30:
Also tried this:
[image]

DanKoloff, 2021-05-04:
Why GPIO5 exactly? Did you toggle GPIO5 in those tests by software means?

Try with one of the GPIOs that are available at the UEXT connector - for example GPIO16 or GPIO13.

iBobik, 2021-05-04:
U use GPIO5 because it is not used anywhere else. Yes, I toggled it by SW.
Is there something wrong with this?

I have not used UEXT pins because I use this connector for something else
(only GPIO 13, 14 and 15).

Honza Pobořil

DanKoloff, 2021-05-05:
Just as a test try with one of the pins from the UEXT instead of GPIO5, for example GPIO16 which seems that you don't use for the other connection.

closed 2021-09-30 by DanKoloff