OLIMEX/ESP32-EVB #31: CAN bus cannot receive anything

githubxup, 2021-05-31:
I tried to receive messages from CAN bus by this board with CAN demo example, but nothing was happened.
I verified that the CAN sender is fine by using arduino UNO + CAN bus shield, and the messages are received as expected.
Do you have any suggestion?
Thank you

James

DanKoloff, 2021-06-01:
1) Make sure that the hardware connections are proper, CAN-HI is next to the power jack, CAN-LOW is in the middle.

2) Try another software, there is default CAN driver in ESP-IDF just change the pins to the ones we use.

3) In ESP32-EVB, the default the termination resistor is disconnected. Try enabling the termination resistor, it is actually required to have the termination resistors enabled in first and last node of a CAN network. If you have only two CAN devices you need to have termination circuits of both. To enable the termination resistor - close jumper CAN_T1 - solder carefully its pads together to establish connection.

githubxup, 2021-06-01:
Hi DanKoloff,

I got it work after sodering CAN_T1.
Thank you a lot.

James

closed 2021-06-01 by githubxup