OLIMEX/ESP32-EVB #34: ESP32-EVB vs ESP32-EVB-IND

nagyrobi, 2021-12-27:
I ordered an ESP32-EVB-IND but I think the mailman brought an ESP32-EVB. Is there any clear notation difference on the PCB that this actually is an IND version? It only has printed ESP32-EVB on it.
The first thing I noticed is that the relays are black and not orange as in the product photo on the website (RECOY LEG.5), this doesn't seem to comply with the -40+85C rated for the IND version board. That's what made me wonder, could it be that by mistake I did not receive an IND version?

I plan to use this board in harsh environment, I specifically need an IND version.

DanKoloff, 2021-12-28:
All ESP32-EVB board variants use same PCB hence white print is the same. However, the Ethernet chip is different modification. One is suitable for wider temperature range. The regular version uses LAN8720A while the industrial uses LAN8720Ai.

nagyrobi, 2021-12-29:
Like this?
[image]

DanKoloff, 2021-12-30:
Yes, my bad. It is 8710A and 8710Ai. This is also visible in the schematic. I've edited my previous post. The chip on your board is the industrial version.

Still I understand why it is not clear, this doesn't seem to be mentioned on the schematic. I will speak to the designers to generate Bill of Materials (BOM) for each board variant so I can upload it to GitHub.

The ESP32-POE and ESP32-POE-ISO have a lot of differences in the design to galvanically isolate the power supply from the POE and from the USB hence they have different PCBs. Still the IND versions just have few different components.

nagyrobi, 2021-12-30:
It's just the question of how can the customer easily identify, check that he received the correct product.

closed 2021-12-30 by nagyrobi