OLIMEX/ESP32-EVB #41: ESP32-EVB_Ethernet_v4.4 compilation error

itsterit, 2023-01-30:
I'm using an ESP32-EVB board and want to run a standard ethernet example.

[image]
[image]
[image]

DanKoloff, 2023-01-31:
Did you try this instead: https://github.com/OLIMEX/ESP32-EVB/tree/master/SOFTWARE/ESP-IDF%20examples/ESP32-EVB_Ethernet_v4.4 ?

DanKoloff, 2023-01-31:
Also you are using ESP-IDF 5.0 for a project clearly labeled as suitable for ESP-IDF 4.4?

closed 2023-04-10 by itsterit