OLIMEX/ESP32-EVB #51: Lipo Battery with CAN and Ethernet

rvxfahim, 2023-10-13:
Does Ethernet and CAN work with a lipo battery ?

DanKoloff, 2023-10-16:
Ethernet works on battery fine, no problems there. The things that don't work on battery are the CAN and the relays.

CAN - it won't work, the CAN driver operates from 4.5V, it is possible to modify the board but this requires placing another CAN driver and modifying a 3.3V/5V SMT jumper.

Also the relays won't work when powering the board from a battery.

rvxfahim, 2023-10-16:
I do not need the relays, so if i short the 3.3V/5V SMT jumper then will CAN start working ?

DanKoloff, 2023-10-18:
No. The on-board CAN driver chip SIT1050 doesn't work at 3.7V, hence I wrote "this requires placing another CAN driver and modifying a 3.3V/5V SMT jumper."

You would also need to desolder the SIT1050 chip and solder in its place 3.3V CAN chip. In the schematic at the bottom of the CAN part there are options listed. SN65HVD232D is 3.3V option but anything in similar package can do, make sure it works at 3.3V, most CAN drivers require 4.5V+ but there are lower voltage options too.

closed 2023-10-20 by DanKoloff