OLIMEX/ESP32-EVB #56: Question about the SD card's CS connection

fredlcore, 2024-02-10:
I'm a bit puzzled about the connection of the chip select signal of the EVB's SD card reader:
[image]

My understanding of SPI is that CS is an active low signal that is to be set to logical low if the device is being accessed. Here, however, it is set to 3.3V constantly (through a 2.2k resistor). I read that if a device is the only SPI device, then CS can be tied to GND to have the device active all the time, but I don't understand how this is supposed to work here.

Is this therefore an error in the schematics or is it supposed to work that way and it's just my understanding of SPI that is wrong?

Thank you!

TsvetanUsunov, 2024-02-10:
this is not CS but CD pin and should be pulled up

closed 2024-02-10 by TsvetanUsunov

fredlcore, 2024-02-10:
Thank you for your prompt and helpful answer, @TsvetanUsunov!