OLIMEX/ESP32-EVB #58: Lag in mqtt/ethernet communication

programmeddeath1, 2024-07-08:
I am using the olimex esp32-evb-ind to execute the position,velocity and acceleration of a robotic arm. The path is generated on a cpu and the coordinates are then sent to the esp32 over ethernet cable.

When sending data over a control cycle of 0.04 (1 point generated every 0.04s or 40ms). The communication time from the cpu to esp32 is avg around 4ms (RTT) ie 2ms for communication
<google-sheets-html-origin><style type="text/css"></style>
ID | sentTime | ReceiveTime | ExecutedTime | Communication time | Execution time
-- | -- | -- | -- | -- | --
1 | 1720436672 | 1720436672 | 1720436672 | 4.03 | 4.96
2 | 1720436673 | 1720436673 | 1720436673 | 3.72 | 4.54
3 | 1720436673 | 1720436673 | 1720436673 | 4.01 | 4.96
4 | 1720436673 | 1720436673 | 1720436673 | 3.5 | 4.37
5 | 1720436673 | 1720436673 | 1720436673 | 4.19 | 5.49
6 | 1720436673 | 1720436673 | 1720436673 | 3.24 | 4.32
7 | 1720436673 | 1720436673 | 1720436673 | 3.62 | 4.5
8 | 1720436673 | 1720436673 | 1720436673 | 3.87 | 4.64
9 | 1720436674 | 1720436674 | 1720436674 | 4.27 | 5.19
10 | 1720436674 | 1720436674 | 1720436674 | 4.27 | 5.17

But if i reducee the control cycle to below 0.03, eg at 0.01, there seems to be significant lag and pile-up of messages causing constant lag
<google-sheets-html-origin><style type="text/css"></style>
Test 1 | sentTime | ReceiveTime | ExecutedTime | Communication time | Execution time
-- | -- | -- | -- | -- | --
1 | 1719317147 | 1719317147 | 1719317147 | 6.15 | 14.68
2 | 1719317147 | 1719317147 | 1719317147 | 14.63 | 25.86
3 | 1719317147 | 1719317147 | 1719317147 | 25.57 | 26.35
4 | 1719317147 | 1719317147 | 1719317147 | 25.49 | 26.54
5 | 1719317147 | 1719317147 | 1719317147 | 26.25 | 27.66
6 | 1719317147 | 1719317147 | 1719317147 | 25.65 | 26.82
7 | 1719317147 | 1719317147 | 1719317147 | 25.97 | 27.35
8 | 1719317147 | 1719317147 | 1719317147 | 25.17 | 26.31
9 | 1719317147 | 1719317147 | 1719317147 | 24.14 | 25.3
10 | 1719317147 | 1719317147 | 1719317147 | 24.15 | 25.05

I tried controlling the background processes by creating the mqtt callback task to pin to a core using xTaskCreatePinnedToCore. Such that it was run on core 0 as well as core 1, but the lag seems to remain.

Also the lag seems to be more on control cycle of 0.02 and 0.03 - <google-sheets-html-origin><style type="text/css"></style>
1 | 1720436867 | 1720436867 | 1720436867 | 3.12 | 28.74
-- | -- | -- | -- | -- | --
2 | 1720436867 | 1720436867 | 1720436867 | 28.88 | 55.37
3 | 1720436867 | 1720436867 | 1720436867 | 55.61 | 56.43
4 | 1720436867 | 1720436867 | 1720436867 | 55.33 | 56.06
5 | 1720436867 | 1720436867 | 1720436867 | 52.8 | 53.77
6 | 1720436867 | 1720436867 | 1720436867 | 47.12 | 48
7 | 1720436867 | 1720436867 | 1720436867 | 47.66 | 48.47
8 | 1720436867 | 1720436867 | 1720436867 | 47.18 | 47.99
9 | 1720436867 | 1720436867 | 1720436867 | 47.01 | 47.81
10 | 1720436867 | 1720436867 | 1720436867 | 47.2 | 48.27

I tried changing the message data size, sending in json format or as encoded byte format, but there seems to be almost no difference in the processing time.

Is this a limitation with the cpu processing capabilities of the esp32 that its not able to process points at such high frequencies.

The esp32 DOIT kit which runs WROOM-32 is able to achieve faster times with uart at a baud rate of 4000000. It seems counterintuitive that uart is faster here than ethernet or mqtt over LAN.

I have connected the esp32 over a link-local only lan setup and no router, the ping times are around 0.3ms.

DanKoloff, 2024-07-08:
This can likely be improved by software means but I doubt this is the proper place you'd receive the needed attention. Better check the communities with the MQQT or ESP32 libraries that you use.

There should be something like TCP_NODELAY in the mqqt options or maybe you can try UDP instead.

"The esp32 DOIT kit which runs WROOM-32", well ESP32-EVB also runs the same module (either ESP32-WROOM-32E or ESP32-WROOM-32UE depending if it has antenna).

>  It seems counterintuitive that uart is faster here than ethernet or mqtt over LAN.

Why? Which interface is easier to process and has less overhead? Ethernet requires huge software stack and UART is much simpler...

programmeddeath1, 2024-07-09:
I understand it has less overhead, but mqtt and ethernet are said to be faster based on all reference content available online.  And it is true too, since at control cycles of 0.04, mqtt takes around 4ms avg for round trip which is way less than usb serial's best time(which is around 8 to 10ms avg). This lag is when messages are getting piled up at a rate of 10ms(cc- 0.01) and there is a constant or increasing delay in their subsequent messages.  

I will try with nodelay and UDP once and check. I'll have to check with the expressif communities, the mqtt and library communities are quite inactive.

Thanks for your response!

closed 2024-07-09 by programmeddeath1