OLIMEX/ESP32-EVB #59: GPIO0 is being triggered at startup?

Arcitec, 2025-06-06:
Hi! :) I'm looking at the latest schematic:

https://github.com/OLIMEX/ESP32-EVB/blob/master/HARDWARE/REV-K1/ESP32-EVB_Rev_K1.pdf

- It uses a 50 MHz crystal.
- The crystal is wired with +3.3v rail going to its VCC and ENABLE pins, meaning that the crystal is *always* running immediately at board power-on.
- The crystal output goes to the LAN8710A and to the ESP32's GPIO0 pin.

Now that's a red flag for me:

- GPIO0 is the pin that tells the ESP32 to go into bootloader/programming/flashing mode.
- So the crystal would be putting a signal on GPIO0 at startup, thus forcing the ESP32 into flashing mode.

Am I missing something?

Actually, just as I was writing this down, I guess that I'm also seeing some kind of oscillator disable/enable circuitry that is wired to the ESP32 Enable pin, but I really don't see how this board works?

DanKoloff, 2025-06-06:
Yes, due to lack of free GPIOs in ESP32-EVB's design, we couldn't sacrifice another pin for ETH enable so we used own circuit to delay GPIO0 by 500ms. Just look around the schematic and the text.

All other ESP32 boards that we have with Ethernet use the standard dedicated enable pin. Like if you compare ESP32-EVB's and ESP32-GATEWAY's schematics around the Ethernet you'd notice the differences.

Arcitec, 2025-06-06:
@DanKoloff Ah I see. So in this design your Ethernet chip is always enabled, and the crystal is controlled by the ESP32 enable (EN) signal with a 500ms time delay to prevent the crystal startup until after the ESP32 has booted already.

That 500ms delay seems very high since the crystal should be started before any ESP32 software tries to talk to the Ethernet chip. But I guess the ESP32 takes longer to boot than I thought.

---

I looked at the ESP-GATEWAY but it doesn't use an external crystal. It uses GPIO17 50MHz clock output from revision D and onwards (A-C used the time delay and external oscillator):

https://github.com/OLIMEX/ESP32-GATEWAY/blob/master/HARDWARE/Hardware%20revision%20I/ESP32-GATEWAY_Rev_I.pdf

---

Unfortunately there is a very severe hardware bug in every ESP32: The 50MHz clock goes unstable and crashes/hangs the ESP32 if you enable WiFi and Ethernet at the same time:

https://docs.espressif.com/projects/esp-chip-errata/en/latest/esp32/03-errata-description/esp32/clock-rmii-clock-when-using-ethernet-and-wifi.html

So I am going to solve it with a LAN8720A and an external crystal. That chip can be clocked with a 25MHz crystal and then generates 50MHz internally. The 50MHx is then output via it's REFCLKO output and needs to be routed into GPIO 0 on the ESP32 so that the ESP32 can talk to the Ethernet chip.

Which means that I need to implement a soft start for the crystal so that it doesn't output a signal during the ESP32 startup process.

Right now it seems like the best idea is to use the ESP32 enable pin signal and time-delay it, like you did.

But it might also be possible to use some other method to enable the crystal. Any ideas? Perhaps it can be tied to one of the Ethernet enable pins.

Arcitec, 2025-06-06:
Oh I think I have found the best solution.

- Wire any GPIO to the crystal enable pin. WT32-ETH01 uses GPIO 16.
- Then specify that GPIO as the "eth_power" pin in WLED. I am not sure if that is an ESP32 SDK feature or a WLED addition. But it's used here (see the WT32-ETH01 power definition here): https://github.com/wled/WLED/blob/main/wled00/network.cpp#L31
- Then you don't need any time delay circuits.

I haven't finished reading through the WT32-ETH01 schematic in detail to see if GPIO16 is connected to anything more or just to the crystal.

And I also need to look up whether GPIO 0 can be directly joined with both the CH340 and REFCLKO traces or if I need some mux/diodes to prevent signals going the wrong way in that trace. :)

Other than those two uncertainties, this seems like an awesome solution.

DanKoloff, 2025-06-06:
I haven't read everything you wrote but there is one important thing that has to be considered and I missed it in my first reply. You need to have the clock available on the LAN8720 chip before it gets powered, else it will hang. "During a Hardware reset, an external clock must be supplied to the XTAL1/CLKIN signal", page 33 of the manual:

https://ww1.microchip.com/downloads/en/devicedoc/00002165b.pdf

DanKoloff, 2025-06-06:
But yes, if you have free GPIO you can just install transistor and hold the Ethernet in reset. Check transistor T1 in ESP32-GATEWAY, its purpose is to hold the LAN in reset during initialization for some milliseconds, also can be used a mean to reset only the Ethernet if it hangs (and not reset the whole board).

Arcitec, 2025-06-06:
Thanks! I will see how the WT32-ETH01 achieved that, since they use GPIO16 to turn on the crystal and their code is at:

https://github.com/arendst/Tasmota/blob/f3d4b5275d55095f6652f3849f67732b7cc71adf/tasmota/tasmota_xdrv_driver/xdrv_82_esp32_ethernet.ino#L232-L238

And in WLED their GPIO16 is listed as the Ethernet power pin, so I will ensure that WLED does the right thing and activates the crystal before trying to talk to the chip:

https://github.com/wled/WLED/blob/main/wled00/network.cpp#L31

I think it sounds like it's safe to power the LAN8720A without a crystal though (making it hang), and then just enable the crystal and send the Reset signal to restart the 8720A.

Arcitec, 2025-06-06:
Thanks for the tip about transistor T1 and holding the LAN8720A in reset.

The final thing that I am really unsure about is the GPIO0 mixing of CH340 signal + crystal clock signal. Is it safe to join those two traces to the same pin?

Or do you need some diode or something to protect the CH340/Crystal frommreverse voltage?

DanKoloff, 2025-06-06:
We use Schottky diodes between CH340 outputs and ESP32 GPIOs, the CH340T is powered from 5V DC while GPIOs of the ESP32 are 3.3V. Like D7, D8, D9, U9, U10.

Arcitec, 2025-06-06:
By the way, just in case it got lost in the discussion, I'll alert you again that the ESP32 has an unfixable hardware bug which is exactly why I won't use its own internal clock anymore. So you might want to consider switching to external crystals again. :/

> Unfortunately there is a very severe hardware bug in every ESP32: The 50MHz clock becomes very unstable and crashes/hangs the ESP32 if you enable WiFi and Ethernet at the same time:
> 
> https://docs.espressif.com/projects/esp-chip-errata/en/latest/esp32/03-errata-description/esp32/clock-rmii-clock-when-using-ethernet-and-wifi.html
> 
> The Ethernet chip produces nonsense if the clock is unstable. And the ESP32 then freezes for up to 30 seconds while it tries to decode the nonsense.
> 
> Which means you can't use ESP-NOW (WiFi) remotes on Ethernet boards unless you install an external crystal, to avoid using the buggy, on-board ESP32 50 MHz PLL.

DanKoloff, 2025-06-06:
Thanks for the headsup. Not much we can do about that at the moment but will see in future. But still WIFI and Ethernet can be used with time-sharing, so it depends on the application I've used both in the past (I guess not simultaneously but...):

https://github.com/OLIMEX/ESP32-POE/blob/master/SOFTWARE/ARDUINO/ESP32_PoE_Ethernet_WIFI_test/ESP32_PoE_Ethernet_WIFI_test.ino

I know customers have used both in the past too:

https://github.com/zorce/ESP32-POE_MQTT_example/blob/master/ESP32-POE_MQTT.ino

Anyway, glad to help, feel free to use our work as basis.

Arcitec, 2025-06-06:
@DanKoloff Thanks a lot again for the advice and explanation of your design, since it's really been helpful in figuring out solutions for using external crystals. :)

I don't think the timesharing solution there is correct. The only way to prevent the hardware bug is to call `WiFi.disconnect()` to turn off the radio.

Any WiFi data (even access point beacons) will disrupt the ESP32's PLL clock signal generation.

That disruption in turn leads to major packet loss and even the ESP32 hanging (if it tries to talk to the ethernet chip).

By the way, thinking about my plan above again, it seems I will need 3x GPIOs:

- GPIO 0 to receive the clock.
- GPIO X (let's say 5) to send the nRST signal.
- GPIO Y (let's say 17) to send the permanent crystal enable/power signal.

I can't think of any ways to combine the "crystal enable" with any other Ethernet pin purposes.

So either I'll have to do what you did with the delayed crystal startup which ensures the crystal is always available at startup and won't waste a GPIO (but needs some complexity with extra parts for the time-delay chip), otherwise I'll have to implement a separate GPIO pin that holds the crystal enable pin high, and add code for that in WLED (something like a `eth_crystal_enable` GPIO variable). 🤦‍♂

Arcitec, 2025-06-08:
@DanKoloff Hi again. I found an official reference now:

https://docs.espressif.com/projects/esp-techpedia/en/latest/esp-friends/advanced-development/performance/reduce-boot-time.html

> The startup time of ESP chips refers to the time it takes for the ESP chip to power on and execute the app_main function. The unoptimized startup process usually takes a long time (generally around 300 milliseconds).

So by delaying the crystal startup by 500 milliseconds, you're delaying it way past the code execution stage of the ESP32 firmware (at 300 milliseconds).

I mentioned that I thought 10-100 milliseconds was a more appropriate value for the crystal delay.

Was there any reason why 500 milliseconds of crystal startup delay was chosen? :)

DanKoloff, 2025-06-09:
It was empirically tested. But it is easy to change as you'd like simply pick lower value for C32 (currently 1uF).

Important notice - the delay is performed by R52, C32, U11. Notice that U8 is NA - not assembled.

Arcitec, 2025-06-09:
@DanKoloff Yeah, I have no doubt that 500ms has been tested across many board revisions and is working. :)

I was just wondering/worried about it since it seems way too high, considering the ESP32 boot sequence:

- 0ms: Power on the ESP32.
- 300ms: ESP32 has finished booting and is now executing the user firmware code. Where the firmware can do things like `ETH.begin()` very early.
- 500ms: Crystal turns on.

That's why I was asking if there was any reason why such a high value was chosen. Perhaps you discovered that "100ms doesn't work" etc?

I think the reason why 500ms works despite being very high is most likely that the ESP32 and LAN87X0A chips both simply hang/freeze when they try to communicate, and simply hang (waiting for the clock) until the clock pulse begins, so having a too-long delay probably doesn't harm anything.

It sounds reasonable to use something like 100-300ms crystal startup delay though, to be sure it's enabled before the ESP32 begins executing user code. Even 50ms sounds like it should work, since the boot mode selection happens within the first few milliseconds of the ESP32 power-on, and that's really the only clock signal interference that needs to be avoided at bootup.

So that's why I am very curious: Was there any test with a lower delay? If so, were there any issues?

DanKoloff, 2025-06-10:
Yes, empirically tested means "trial and error".

Arcitec, 2025-06-10:
@DanKoloff Thank you, I highly appreciate all the testing you have done to make sure that 500ms is rock solid!

The reason I was asking if you tried lower values is because the ESP32 manual says the following:

- After the voltage is high enough to turn on the chip, it takes around 3-4ms for it to start reading the GPIO 0 value.
- It then takes it 1ms to read the GPIO 0 value to choose the boot mode.
- After the bootstrap setting has been read, the GPIO 0 pin becomes a "regular I/O pin". Meaning it's safe to start blasting it with the clock signal.
- So, **4 or 5 milliseconds after boot, it's safe to start the crystal.** That's how fast the GPIO 0 becomes a regular I/O pin at boot.
- I definitely listened to your testing though, and have decided to use a 200 millisecond delay personally. It means it's definitely fully started before the ESP32 begins executing user code (at 300 milliseconds), and it's still long enough that it'll never be a problem.

[image]